robinhedwards/A8PicoCart #2: Need for removing cartridge when plugging USB cable

jhusak, 2023-10-01:
In the manual:
*ALWAYS REMOVE THE CARTRIDGE FROM YOUR ATARI BEFORE PLUGGING IN A USB CABLE*

What could happened? Was it tested and something happened? How could (cheap way) be this handled not to remove cartridge?
Removing and inserting cartridge many times is harmful either for the slot or for cartridge itself.

And in fact not the cable is a problem but adding new 5V power source from the host computer.

robinhedwards, 2023-10-01:
The risk is the USB port back-powering the Atari through the cartridge
port. There is a diode on the purple pico clone which prevents the opposite
happening. Although I haven't tested it, I would say the risk is real and
damage is likely to occur.

There is a fuller discussion of the whole issue on the AtariAge thread, and
suggestions as to how the board could be modified to remove this risk.
https://forums.atariage.com/topic/351546-a8picocart-unocart-on-a-raspberry-pi-pico/

I have solved the issue in the next version of the PCB by rotating the pico
through 90 degrees such that the USB port is facing downwards. This makes
it physically impossible to plug in USB when the cartridge is inserted.

The atari cartridge port is designed to have cartridges plugged in and
removed constantly, so I'm not worried about that. The wear to the
cartridge itself is more of a concern, I agree - but this is designed to be
a cheap and simple device.

New version of the PCB will be posted on github shortly.

jhusak, 2023-10-01:
I think a diode from Atari 5V to pico can prevent any issues. Think of it as developing tool. You develop, upload, test. Without removing a cart. I think it is doable, or maybe I'm missing something?

robinhedwards, 2023-10-02:
Have a look at my posts on page 6 of the thread I linked on AtariAge.
A diode is not enough.

Robin

jhusak, 2023-10-02:
So there is a try with hct4053 :) I'll watch the topic just in case.

robinhedwards, 2023-10-07:
New version of the PCB uploaded today makes it physically impossible to plug in USB cable when cartridge is inserted, so I'm going to mark this as resolved.

closed 2023-10-07 by robinhedwards