robinhedwards/A8PicoCart #11: rare carts support

viktorcech, 2024-01-20:
hello, its possible add funcion to run ROM carts which have button to load? some old czechoslovakia cartridges works like that

ascrnet, 2024-03-10:
Hello I made an implementation of a timer to simulate pressing a button, it was made for turbo 2000. I attach forum thread. 
https://forums.atariage.com/topic/351546-a8picocart-unocart-on-a-raspberry-pi-pico/page/19/#comment-5396027
Greetings

viktorcech, 2024-03-10:
hi, i check it, thanks
its not for turbo, but for carts like theese

viktor čech
http://turiecfoto.sk

viktorcech, 2024-03-10:
here is a video:

https://www.youtube.com/shorts/MaE52GzUbwU

ascrnet, 2024-03-15:
Here you will find a compatible one 

https://forums.atariage.com/topic/351546-a8picocart-unocart-on-a-raspberry-pi-pico/page/22/#comment-5429923