robinhedwards/A8PicoCart #13: Documentation: Specify thickness / other features of the pcb?

BrainSlugs83, 2024-03-04:
Hey,

I don't really know how to design PCBs, so maybe some of these things are specified in the file already, but usually on multi-cart projects they include information regarding what you need to specify when you order the PCB...

I'd like to order a batch of these from JLCPCB, but I have some questions that I don't see specified anywhere (at least not in English in any of the markdown files):

- Gold Fingers required? (Assuming yes; 1U?)
- Does the edge connector need beveled connections ("30 degree fingers chamfered")?  (Assuming yes)
- What is the thickness of the PCB? (seems the default is 1.6mm on JLCPCB, is that correct for this? -- or did it pick that up from the gerber file? -- is there a way I can check myself?)
- Minimum Trace Spacing?
- Castelated Holes? (Assuming no)
- Edge Plating? (assuming no)
- Any special notes we need to provide?

robinhedwards, 2024-03-04:
Agreed, I'll add this information to the PCB files readme.

In the meantime - the main thing to get right is PCB thickness of 1.6mm

Gold fingers and bevelling not required (but both would be good and aid longevity).

Robin

robinhedwards, 2024-03-04:
added thickness to pcb readme

closed 2024-03-04 by robinhedwards