robinhedwards/A8PicoCart #14: suport atari 2600 ?

manotroll, 2024-03-16:
suport atari 2600 ?

sail2themoon, 2024-06-03:
I have the very same question, and can't find an answer.. I've seen an image of a "2600 Picocard". I'm quite new to this, because I've just repaired an Atari 7800. Pls help :)

TheVintNerd, 2024-06-03:
Hi. Hopefully I can help. I did not create this project at all, but I did just post a YouTube video of soldering one together and trying it out: https://www.youtube.com/watch?v=A8cQarAvKoU

This project, the A8PicoCart, is Only for the cartridge slot of Atari 8-Bit computers (400, 800, 1200XL, 600XL, 800XL, 65XE, 130XE, 800XE, XE GS (Game System)). This project's cart will Not work with the Atari VCS/2600, 5200, 7800, etc.

Hope that helps! -Cheers, Steve aka The VintNerd on YouTube

sail2themoon, 2024-06-03:
Hi Steve,
thank you very much for bringing some light into this. Then I'll look for the 2600-version. It exists as a photo in the Atari-forum at least.

Cheers
Simon