robinhedwards/A8PicoCart #16: max number of roms files on the pico ?

pepouille, 2024-07-10:
Hi,

Is there a maximum number of files in the filesystem on the pico (like) device ?
I've been able to copy only 110 files on it, while the available space is still still huge (+/- 13 MB).

robinhedwards, 2024-07-10:
File storage is FAT, so there shouldn't be a practical limit to the number of files on the filesystem itself. However the A8PicoCart firmware itself will stop reading the files in a directory (for the atari menu) after 255 entries, so you need to organise files in directories if you want more than 255 files on the device.

If you are having trouble with the USB file storage, try resetting the filesystem as described in the manual?

closed 2024-07-10 by robinhedwards

pepouille, 2024-07-10:
yes, I've reset the board, and rearranged the files in directories, and it solved the problem I had indeed.
Thanks for your answer, and for the achieved work on this project ! ;-)