robinhedwards/A8PicoCart #17: Atari 8bit ram tester as NO_OS.ROM

goloskokovic, 2024-10-06:
Would it be possible to use Shoestring’s Atari 8bit ram tester instead of OS ROM?
https://jammarcade.net/shoestrings-atari-8bit-ram-tester/

This RAM test is 16k rom to be burned to 27c128 and installed in the OS socket (U5) on the Atari motherboard.
By renaming RAM test file to NO_OS.ROM in A8PicoCart can we use it to find bad DRAM?

[image]

robinhedwards, 2024-10-06:
From a brief scan of the code, it might work. Easiest thing to do is to try?

Regards,

Robin

goloskokovic, 2024-10-14:
tried to rename file but it is ignored in root.

how do you jump here?
https://github.com/robinhedwards/A8PicoCart/blob/main/source/Pico%20VSCode%20Project/a8_pico_cart/atari_cart.c#L1630

robinhedwards, 2024-10-14:
Try launching an ATR file. That’s what causes the soft OS to be loaded,
then a reboot should happen into the new OS

goloskokovic, 2024-10-15:
when running random ATR file it starts test but freezes.

for the reference - I was able to run this test in Altirra as OS rom

goloskokovic, 2024-10-30:
I've tried to replace osrom.h with the ram_test.rom using xxd. once I built new uf2 file it behaved just as with uno_os.rom on atr load.
Can I load osrom.h before a8picocart "explorer" like [here](https://github.com/robinhedwards/A8PicoCart/blob/main/source/Pico%20VSCode%20Project/a8_pico_cart/atari_cart.c#L1543) ?

flashjazzcat, 2024-11-07:
The loader turns off the OS ROM before loading the patched Altirra OS, and the patched OS is careful never to re-enable ROM by setting bit 0 of $D301 (PORTB). The RAM test ROM presumably assumes it's in ROM and repeatedly sets bit 0 of PORTB; this will instantly reinstate the OS ROM and cause a crash.

You'd need a special version of your RAM tester designed to run in RAM under the OS.