robinhedwards/A8PicoCart #18: GPIO at 5V

crystalct, 2024-11-05:
The GPIOs of the Raspberry Pico clone thus work at 5V, well above the specific 3.3V +/- 0.5V. Won't it fail in the long run?

pmandes, 2024-11-09:
Hi, Read this: https://hackaday.com/2023/04/05/rp2040-and-5v-logic-best-friends-this-fx9000p-confirms/

nanoant, 2025-01-13:
I'd like to chime in here. According to linked article "Some major points – certain GPIOs (like ADC ones) can’t take it, the GPIOs aren’t 5V-tolerant when set to output, and you shouldn’t feed the GPIOs 5V when the RP2040’s VDDIO is not powered up.".

But I can see we bind both input and output pins to directly to Atari's 5V domain. I guess if Pico fails then that's not a big loss, but what if RP2040 fails by creating short at its power line connected to more precious retro computer? I don't see short circuit protection mentioned in RT6150 SMPS IC used by Pico. Are we absolutely confident that we won't burn Atari when RP2040 fails?

There's also "caveats" link point to some discussion: https://forums.raspberrypi.com/viewtopic.php?p=2092122#p2091977

So what speaks against throwing 3 of TXS0108 other than cost of `$1.25` each `~$4` total and extra complexity in wiring everything up?

nanoant, 2025-11-10:
According to this announcement https://www.raspberrypi.com/news/rp2350-a4-rp2354-and-a-new-hacking-challenge/ leading to this post in comments sections https://forums.raspberrypi.com/viewtopic.php?p=2330278#p2330278 all RP235x Ax are 5V tolerant not just A4, and older RP2040 are obviously NOT 5V tolerant (at least officially).

Then there are several interesting boards using RP2350B with all 48 pins exposed that open interesting possibilities like having SD card slot or e.g. having RAM expansion similar to AVGCART. Some of the boards include 8MB PSRAM and 16MB flash like this one https://olimex.wordpress.com/2025/09/22/new-rp2350-pico2-open-source-hardware-development-board-with-all-48-gpios-exposed-in-breadboard-friendly-layout/ but there are many other RP2350B from Chinese vendors.